TimPaterson/Fusion360-Batch-Post #3: Support for GRBL Post configuration

mateo4778, 2020-10-08:
I hacked together support for the GRBL Post configuration and it appears to work in the one test I ran, but keep in mind this modification completely breaks (ie overwrites) the current Tormach implementation so it would need to be refactored to be properly integrated with the existing code base. I just figured i'd throw it on here in case someone else was in my situation and was looking for a quick/dirty solution to get by on. Also I didn't test the Rapid Z feature with this change so it may or may not work.

Below are the two modified sections of code
```
constEndProgramGcode = "M30\n"
```

```
def PostProcessSetup(fname, setup, setupFolder, docSettings):
    ui = None
    fileHead = None
    fileBody = None
    fileOp = None
    retVal = PostError.Except

    try:
        app = adsk.core.Application.get()
        ui  = app.userInterface
        doc = app.activeDocument
        product = doc.products.itemByProductType(constCAMProductId)
        cam = adsk.cam.CAM.cast(product)

        # Create PostProcessInput
        opName = fname
        if docSettings["splitSetup"]:
            opName += constOpTmpFile
        postInput = adsk.cam.PostProcessInput.create(opName, 
                                                    docSettings["post"], 
                                                    setupFolder, 
                                                    docSettings["units"])
        postInput.isOpenInEditor = False

        # Do it all at once?
        if not docSettings["splitSetup"]:
            try:
                if not cam.postProcess(setup, postInput):
                    return PostError.Fail
                time.sleep(constPostLoopDelay) # files missing sometimes unless we slow down (??)
                return PostError.Success
            except:
                return PostError.Except

        # Split setup into individual operations
        path = setupFolder + "/" + fname
        pathlib.Path(setupFolder).mkdir(parents=True, exist_ok=True)
        fileHead = open(path + constGcodeFileExt, "w")
        fileBody = open(path + constBodyTmpFile + constGcodeFileExt, "w")
        fFirst = True
        lineNum = 10
        fFastZenabled = docSettings["fastZ"]
        if fFastZenabled:
            strReg = (r"(G(?P<G>[0-9]+)[^XYZF]*)?"
                "(X(?P<X>-?[0-9]+(\.[0-9]*)?)[^XYZF]*)?"
                "(Y(?P<Y>-?[0-9]+(\.[0-9]*)?)[^XYZF]*)?"
                "(Z(?P<Z>-?[0-9]+(\.[0-9]*)?)[^XYZF]*)?"
                "(F(?P<F>-?[0-9]+(\.[0-9]*)?)[^XYZF]*)?")
            regGcode = re.compile(strReg)

        i = 0;
        ops = setup.allOperations;
        while i < ops.count:
            op = ops[i]
            i += 1
            if op.isSuppressed:
                continue

            # Look ahead for operations without a toolpath. This can happen
            # with a manual operation. Group it with current operation.
            # Or if first, group it with subsequent ones.
            hasTool = op.hasToolpath
[... 175 lines cut ...]
            try:
                fileHead.close()
                os.remove(fileHead.name)
            except:
                pass

        if fileBody:
            try:
                fileBody.close()
                os.remove(fileBody.name)
            except:
                pass

        if fileOp:
            try:
                fileOp.close()
                os.remove(fileOp.name)
            except:
                pass

        if ui and retVal != PostError.BadFormat:
            ui.messageBox('Failed:\n{}'.format(traceback.format_exc()))

        return retVal
```

TimPaterson, 2020-10-08:
I'm really interested in making this work as widely as possible, so I'll need to figure out a common ground.  Is G30 bad for you?  Obviously I'm not an expert on G-code, just familiar with my Tormach.

Can you upload a sample G-code file so I can understand it better?

mateo4778, 2020-10-09:
I also only have a practical level of experience with g-code working with my cnc mill that uses the grbl firmware. But also I think you could investigate the other types by making a simple model in fusion 360 and then generating the post g-code using different post output types (like grbl). That is probably the best way to compare the output differences. 

I am not necessarily sure if the G30 is a problem, I just see that it is not generated by the grbl post processor, and my goal was to have the end file ideally replicate what the post processor would have put out. In striving towards a more generic solution I think the best approach would be to detect the relevant information as generically as possible, and then blindly passing everything else through to the final output file, that way you don't have to care about the specific preferences of one machine or another, and just let the post processor worry about that. 

Basically comparing a bunch of g-codes from different post processors and then identify the (hopefully) common formatting aspects, and using those aspects to parse the files. 

Below is the relevant bits from one of my g-codes output from the fusion 360 grbl post processor. Note the differences from the Tormach output. It doesn't have the "%" at start and finish, it doesn't use the "N10" line numbering at the start of the operation, and it doesn't use G30 at the end of the program. So you can see the obvious conflict because the current logic in PostProcessSetup was explicitly relying on the "%" at start and finish and the "Nxx" line identifier at the start of the operation.

In my modified version of PostProcessSetup I used the M30 to identify the end of the g-code, and I used the tool change "T" operation to identify where the program starts, assuming the op title line was right above that. I don't necessarily know how robust/reliable those two relationships are beyond these few samples, but at least those two things appear to be common between the Tormach and grbl outputs, so maybe it's an improvement. (Note I didn't test my changes with a Tormach g-code but it should work for both with maybe some small tweaks since the two relationships are common)

```
(ProjectTitle)
(this is a comment)
(T3  D=0.25 CR=0 - ZMIN=-0.375 - flat end mill)
G90 G94
G17
G20
(When using Fusion 360 for Personal Use, the feedrate of rapid moves is reduced to match the feedrate of cutting moves, which can increase machining time. Unrestricted rapid moves are available with a Fusion 360 Subscription.)

(Pocket1)
T3 M6
S20000 M3
G54
G0 X8.344 Y2.0105
Z0.6
G1 Z0.1394 F60
Z0.1237 F30
X8.3441 Y2.0111 Z0.1183 F50
X8.3447 Y2.0128 Z0.1131
X8.3456 Y2.0155 Z0.1084
X8.3469 Y2.019 Z0.1045
....
....
....
X4.256 Y4.1179 Z-0.3608
X4.2571 Y4.1164 Z-0.3556
X4.2575 Y4.1159 Z-0.35
Z0.6
M5
M30

```

TimPaterson, 2020-10-10:
You're right, I can run any of the post processors myself.  Good idea!

I have new, simpler model for the G-code format that now works with both Tormach and GRBL. Give it a try and let me know if it's working right for you. I may also start trying some other post processors, but there's so many it's hard to know which ones to choose.

closed 2020-10-10 by TimPaterson

mateo4778, 2020-10-12:
The latest appears to work well for my use case at least, thanks!