TimPaterson/Fusion360-Batch-Post #5: Long lasting operations are missing GCode

thomam04, 2020-10-15:
I have noticed that Gcode is missing from some of my larger operations (3D Adaptive or 3D Contour). I have compared the Gcode generated by Post Process All vs posting Gcode for single operation.  

It seems like the script is having issues when the generated Gcode is exceeding a certain limit - e.g. 10000 lines of Gcode. I have noticed this primarily with operation that have 10000+ lines of Gcode.

TimPaterson, 2020-10-15:
Are you using the "individual operations" feature needed for the free version (this makes a big difference)?

Just in case, what post processor are you using?

thomam04, 2020-10-15:
Yes, I am using the individual operations feature. The only PP that works for me right now is the Tormach.cps

thomam04, 2020-10-15:
Not sure if this helps but here is a screen shot of VSCodium comparing the GCode.

[image]

[image]

thomam04, 2020-10-15:
I am also getting different Gcode file sizes when I run Post Process All multiple times. In between, I delete the file and post it again. Not getting any error messages, though. 

[image]

[image]

[image]

TimPaterson, 2020-10-15:
I don't have anything big enough to reproduce this, but I'm pretty sure I know what caused the problem. Give it another try now  and see if its fixed -- close the issue if so.

thomam04, 2020-10-16:
Hi Tim, 

unfortunately the problem still persists. You can try it yourself by using a build-in CAM example:

In your project folder go to SAMPLES (all the way at the bottom) -> CAM Samples -> Fusion Keychain. 

Generate Gcode for Setup1 a couple of times and you will notice a size difference. I have compare both files with VSCodium and I can't find a pattern on what it will skip.

TimPaterson, 2020-10-16:
I used the sample but was unable to reproduce the problem.  I generated the operations individually (without my add-in) to verify that all the code for each is present in the combined output.  I have attached the output from the add-in.  Maybe you could attach one of the output files that is incomplete for me to look at, but I'm not sure I'll be able to solve it without reproducing it on my machine.

[Setup1.txt](https://github.com/TimPaterson/Fusion360-Batch-Post/files/5393130/Setup1.txt)

thomam04, 2020-10-16:
I have recorded two videos to demonstrate the problem. Not sure if this helps but at least you can see if I am doing something wrong here. 

https://youtu.be/cKZW8WIBS3k

https://youtu.be/zXVOfHTnBnM

TimPaterson, 2020-10-16:
First, just to be sure, when you updated the add-in you must restart it -- either by restarting Fusion 360 itself or going back to the Add-Ins menu and stopping it, then restarting.

Assuming that's been done and didn't help, how would you feel about doing a little debugging yourself? I'd be looking for certain specific things while the program is running, so it wouldn't require understanding the program.  Basically, there are intermediate files we would want to look at, so we need to stop the program to get a peek before it deletes them.

The basic instructions for debugging are at: 
https://modthemachine.typepad.com/my_weblog/2019/09/debug-fusion-360-add-ins.html
If you think you want to give it a shot, I can give you details of what we need to do. I can also give you an email so we can take this offline and communicate directly.

thomam04, 2020-10-17:
Hi Tim, I am not a programmer but I can give it a shot. If you want to give me your eMail address, we can take this offline. 

Regards,
Markus

TimPaterson, 2020-10-17:
Here is a temporary one: gittemp, with my GitHub user name as the domain, plus dot com.

TimPaterson, 2020-10-21:
This seems to be a timing issue specific to MacBook OS X. No further action planned for now.

closed 2020-10-21 by TimPaterson