TimPaterson/Fusion360-Batch-Post #6: Is this add-in working for you?

TimPaterson, 2020-10-21:
I have it working with my Tormach mill.  (It's been tested with many other post processors, but the resultant G-code has only been reviewed and not actually run.)

Add a comment on what machine you've tried it on.

ryanjrose, 2020-10-22:
I'm using Fusion360 v2.0.9142
First, I cannot find an option to "Run on Startup".   When I click on Run, nothing happens.  I check the Actions menu and the only items I see are: Automatic In-Process Stock Generation, Simulate, Post Process, Setup Sheet

TimPaterson, 2020-10-22:
[image]
This is an image of the Fusion 360 add-in dialog.  Tell me how yours is different.

ryanjrose, 2020-10-22:
Well, I see my error. I added it on the Scripts tab despite your
instructions saying to choose the Add-ins tab. Sorry, my mistake.
Unfortunately, now when I try to add it to the Add-Ins tab nothing happens
when I browse to the folder I git cloned to.  Perhaps because it's
registered as a script already?  I tried to right-click and delete/remove
the script add-in but there is no right-click context menu that allows me
to do that (despite reading about it).  SO, I next tried removing the
directory that I got cloned the add-in to and rebooted Fusion 360.  That
removed it from the scripts tab. but I'm still unable to add it to the
add-ins tab.

geidnert, 2020-10-22:
Hi! I love this initiative, great work! Unfortenately i cant load the addin at all, when i choose the dir where i downloaded it to, i only have 1 file called PostProcessAll.py and in the installation instruction you are talking about "all" files PostProcessAll.*
Dont know if that is the problem that i only have 1 file and the resources folder.

Selecting the dir from the green plus sign in the addin section:
[image]

When the selection window closes it goes back to this, no messages no nothing, nothing in the list:
[image]

Regards Andreas

thomam04, 2020-10-22:
@geidnert Please see this Autodesk guide on how to manually install an add-in. 

https://knowledge.autodesk.com/support/fusion-360/troubleshooting/caas/sfdcarticles/sfdcarticles/How-to-install-an-ADD-IN-and-Script-in-Fusion-360.html

A script belongs in the Scripts folder. An add-in belongs in the AddIns folder.

On Windows, these are the folder paths:

For a script, %appdata%\Autodesk\Autodesk Fusion 360\API\Scripts
For an Add-In, %appdata%\Autodesk\Autodesk Fusion 360\API\AddIns

geidnert, 2020-10-22:
Thx, that worked, moved folder i had on desktop to that addin folder, and then when i pressed the green + on windows i saw the dir, clicked it and it was in the list and i could start it anc choose to run it at startup, perfect! lets se how this performs now :)

Maybe you could add that part in the installation instructions in the readme file.

/A

TimPaterson, 2020-10-22:
I'm so sorry -- there was also supposed to be a PostProcessAll.manifest file that tells Fusion 360 about the add-in.  When I tried a new installation of the add-in, I also saw the silent failure at first, but it works once the manifest is included.  You can install from any folder with the manifest.

footpetaljones, 2020-10-23:
I was able to install the add-in without issue, but I'm not able to get it to post code correctly. I'm using the Dynapath post, which has a strange code format that is probably the cause of the issue. Image of the error below. 

[https://imgur.com/a/lMauJIL](url)

thomam04, 2020-10-23:
@footpetaljones Make sure that you match your Post Processor's extension ".txt" with the Plug-in output file extension. By default it is set to ".nc" - change it to ".txt" which is the correct setting for your PP and see if it will post your Gcode.

footpetaljones, 2020-10-23:
> @footpetaljones Make sure that you match your Post Processor's extension ".txt" with the Plug-in output file extension. By default it is set to ".nc" - change it to ".txt" which is the correct setting for your PP and see if it will post your Gcode.

I do have the extension set to .txt

TimPaterson, 2020-10-23:
The Dynapath output is quite different from normal G-code. However, I made a couple of changes and seemed to get working results. I'm concerned it could impact other post processors, so I'm keeping it separate. On GitHub, that means a new branch called Dynapath.

So if you kept it as a Git repository, you now need to perform a checkout on the Dynapath branch. If you're not into Git, you can use the copy I've attached here:
[PostProcessAll.zip](https://github.com/TimPaterson/Fusion360-Batch-Post/files/5431005/PostProcessAll.zip)
This is also available by selecting the branch from a drop-down on the main page, just above the list of files.

I'm not sure it will work, so you'll need to let me know. It adds a space after each line number; if that's a problem, there's about a half-dozen places that would need to change to take it out. Also, please let me know what tool change G-code to use, if any,.

footpetaljones, 2020-10-24:
With the new version, I no longer get any errors and the code follows Dynapath's conversational format, so it seems that it is working. I always have the property "Use RS274" set to yes so that normal g/m codes are output. If you would be kind enough to add that property, I can test it this weekend. 

Fusion only output a M5 for tool changes before the Tnn, which hasn't been an issue for me, as I don't have a tool changer.

TimPaterson, 2020-10-24:
The add-in uses whatever options are set last time you ran the built-in post process command.  Using the Dynapath post processor on a single operation, I changed the "Use RS274" and "Separate words with space", and ran it once so it would be saved. Then I ran Post Process All and got output that looks more like normal G-code.

As for tool change code, I don't have a tool changer either. But it is nice to stop the coolant and raise the head to home to get some working room -- thus M9 G30 for my Tormach. It stops the spindle on its own, no M5 needed in my case.

footpetaljones, 2020-11-02:
Wasn't able to get out to the machine until today, unfortunately. 

The spaces between the line numbers was an issue, the control wouldn't accept the code with them in. Strangely, the 2d milling cycles are using g0 to get to the retract plane, but any linking moves after the initial move are in g1. That is occurring with the facing operation, 2d adaptive, and 2d contour. Drilling with rapid out is using g0 for each retract as it should, though. Odd. 

I think that having just M9 for a tool change will have to do. I'm not sure how much things will differ for different machines, but at least the Delta 20 control I have can't take the usual G28 or G30 you'd have there.

BenGuse219, 2020-11-02:
Hello TimPaterson,
I was able to use this addin with no problems on 1 of my fusion360 projects. when attempting to use it on another. I had a traceback error. 
[image]
Any idea why it would work on 1 project but not another?

TimPaterson, 2020-11-03:
I'm closing this thread as it has become muddled with multiple issues.

closed 2020-11-03 by TimPaterson