TimPaterson/Fusion360-Batch-Post #7: Do you having it working?

TimPaterson, 2020-11-03:
If you're having any trouble, please create a new issue. This will allow a focused response and easier tracking.

If you have it working, add a comment here. Include the post processor you used, and the settings for the add-in: tool change G-code, file name extension, and if it requires a numeric name.

Muzzerboy, 2020-12-21:
Hi Tim

As noted, the Centroid add-in seems to be working nicely now. I have no "tool change" parameters as they are redundant (already handled by Centroid when a new tool is encountered via the "Tx" statement), although I have the G28 G91 Z0 to give a tool retract at the end of the file. The rapids seem to work well and consistently.

The main issue I have now is that the post processor settings window no longer comes up. I mean this thing:
[image]

It's not unusual to want to change or at least check the settings in the post, yet the add-in seems to suppress the settings window. It may be that it's not easy or possible due to the way you set up the add-in but rather like the g-code popping up in the editor at the end, it's one of those ops that I suspect a lot of people would want.

It seems that I can go into the default post processor, make changes in the settings window, post out a single op and then run process all. Is this a reasonably dependable way to make changes if the settings window can't be brought back? It's a bit of a fiddle but seems to allow us to make changes to the post processor settings at least.

Murray

TimPaterson, 2020-12-21:
The Fusion 360 API does not provide access to the dialog or the settings it keeps.  If you need to change them, you will have to run the built-in post, as you mention.

closed 2020-12-21 by TimPaterson