TimPaterson/Fusion360-Batch-Post #8: Weird error when Posting

Schwarzi07, 2020-11-08:
Every time I try to ron the postprocess I get this weird error:

[image]

I don´t know what is causing this error.I´m using the GRBL Postprocessor.

TimPaterson, 2020-11-08:
Two of your operations have a warning icon on them.  Run the standard built-in post on each of those operations by themselves and resolve any problems.  Let me know what happens.

Schwarzi07, 2020-11-19:
i tried that but the same thing happened.

TimPaterson, 2020-11-20:
I don't know if that means you resolved the warnings on those operations, but in any case I don't have any further suggestions.  The error message means that Fusion 360 did not generate the G-code file for an operation (after five tries).

insolentgamesadmin, 2020-12-11:
I get the same error message.
I do not have any warnings for any operation and I can use the standard post processor for each operation successfully. 
I am using the tomach.cps post processor

[image]

joerobertson, 2020-12-11:
I get the same error message as Schwarzi07.
I do not have any warnings for any operation and I can use the standard post processor for each operation successfully.
I am using the tomach.cps post processor

[image]

joerobertson, 2020-12-11:
I was able to get this to run by changing the location of the Output Folder.
I normally write to a network file share.
I changed Output Folder to a local dir (C:\cncfiles) and it worked.

It appears that in order to work with the network file share, the Output Folder value for the Post Process All dialog
must be the same as the Output Folder entered in the normal Post Process dialog.

I'm not sure why using C:\cncfiles worked without making the Post Process Output Folder the same.
I tried several options and could only get the network share to work if both Output Folder entries were the same.

TimPaterson, 2020-12-11:
The add-in takes starts off with these two steps:
1. Ask Fusion 360 to generate G-code to a temporary file at the location you specify for output.
2. Open the G-code file to process it.

The error message is saying that step 2 failed because the file wasn't there ("No such file or directory").  I'd already found a delay was necessary between step 1 and step 2 -- it retries this 5 times, doubling the delay each time (so 0.1 seconds to 1.6 seconds).

Your discovery about network shares is very helpful. There's no reason for the temporary file to get put in the folder of the final destination.  I will look into revising this to make sure the temporary file is local, while still writing the final results to the desired location.

TimPaterson, 2020-12-11:
I have updated PostProcessAll to use a local folder for temporary files. This just seems like a good idea and I think it will resolve the issue with sending output to a network share. Please give it a try (setting up the condition that was failing, if you can).

Schwarzi07, if you're still listening, you might try it as well.

joerobertson, 2020-12-11:
Tim,

The new version works for local drives (I tried both C:\cncfiles and F:\cnc-gcode)
Found files in the block subdir
I then renamed block:setup to setup and ran again - it worked with the files in the top level directory.

I cannot get the fileshare to work reliably - even if the Output Folders are the same.
I set both Output Folders to \\FILESERVER\tools\Mills and Lathes\Fusion360\post_test_2
It appears to work but the files are in the wrong folder and I end up with the same error message.

I found the results in \\FILESERVER\tools\Mills and Lathes\Fusion360\post_test_2\
but not in the block subdir.
I then renamed block:setup to setup and ran again - it worked.
I cannot see a pattern here - I suspect it may have something to do with the paths for the fileshare.

Is there a reason you leave the 991 Setup.bak file?
It appears to be just the content of the last operation and does not appear to be useful.

TimPaterson, 2020-12-11:
You have the latest version, but you're not using it. You need to stop and restart the add-in or restart Fusion 360.  I know this because: 1) the temp file name would not be "991 Setup" and it wouldn't be looking for it on the remote server; 2) the source line displayed at the location of the error ("time.sleep...") can't cause an error (your previous image shows it properly at a file open()).

joerobertson, 2020-12-12:
oops... I didn't understand about the requirement to restart.
OK - I stopped/started the add-in.
It now works for the fileshare.  It did not work initially but after I ran it three or four times it appears to work now.

Also - I no longer see the .bak file - that is good.

It will be some time before I am able to actually run the gcode to validate that but it looks good.

I am using M5 (to stop the spindle) for G-code to precede the tool change - I saw that in my previous gcode.
  G-code to precede tool change
   M5 M9 G30

Great job Tim!

BlackHawk3000, 2020-12-13:
Good day, 
I also have a problem with the tool.
I get this error as soon as I run it. 
Changing the output path has brought nothing.

What could I test here?
[image]

BlackHawk3000, 2020-12-13:
I changed now the outpufile ending to .ngc an the fail message istnt coming. 
But now i get this message:

[image]

Where can i find the log for bug report?

TimPaterson, 2020-12-13:
BlackHawk3000, your issue has been moved to https://github.com/TimPaterson/Fusion360-Batch-Post/issues/10 as it unrelated to this thread.

TimPaterson, 2020-12-17:
The issue from joerobertson of posting directly to a server seems to be resolved.

The status of the issue from Schwarzi07 is uncertain as no further comments have been received. This could possibly be the same as Issue #10, which is resolved in the latest version by increasing the time delay before checking for post processor output.

closed 2020-12-17 by TimPaterson