TimPaterson/Fusion360-Batch-Post #10: Fail message reported running Post Process All

TimPaterson, 2020-12-13:
I changed now the outpufile ending to .ngc an the fail message istnt coming. 
But now i get this message:

[image]

Where can i find the log for bug report?

Fusion creat the file in the TMP folder and an empty file in the output folder.

LinuxCNC (EMC2) PP

I tried a other PP (tormach), the same issue :(

_Originally posted by @BlackHawk3000 in https://github.com/TimPaterson/Fusion360-Batch-Post/issues/8#issuecomment-744030833_

TimPaterson, 2020-12-13:
This message comes from Post Process All.  It means that Fusion 360 reported that it failed to post one of the operations in the setup.

You should post each operation in the setup individually using the built-in post process command.  I would expect one of them to fail, and Fusion 360 will hopefully give a reason why.

BlackHawk3000, 2020-12-13:
Hey Tim,

you mean this operation?

[image]

If I do that for every operation it works fine, no issues.

BlackHawk3000, 2020-12-13:
So some new information:

- If I use only one tool [T1] with different operations and dont use [Use individual operations] - WORKS FINE
- If I use only one tool [T1] with different operations and use [Use individual operations] - 1/20 or more trys works
- If I more tools [T1,T2...] with different operations and use [Use individual operations] - 1/20 ore more trys works
- If I more tools [T1,T2...] with different operations and dont use [Use individual operations] - 1/20 ore more trys works

TimPaterson, 2020-12-13:
Sorry, what does "1/20 or more" mean?

BlackHawk3000, 2020-12-13:
1 out of 20 or more attempts works and it outputs a complete file.

TimPaterson, 2020-12-13:
So it does work, rarely -- that's very helpful.  Let's try having Post Process All give the operation more time.  This would require you to edit one line of the file PostProcessAll.py.  After making any change, you need to restart the add-in from the Fusion 360 Scripts and Add-Ins dialog, or simply restart Fusion 360.

If you bring up PostProcessAll.py in an editor, you should find "constPostLoopDelay = 0.1" on line 42 (or you can just search for it).  I suggest changing the number to 0.5.  After each try it doubles the amount of time it waits, and it tries 5 times.  That would increase the time allowed for the last try from 1.6 seconds to 8 seconds.  If it works occasionally now, that should give it plenty of time without making you wait too long so see if this idea works (the total time waiting if it fails will be 16 seconds, up from 3.2).

BlackHawk3000, 2020-12-13:
It looks like this was the problem. 
The files are being output without any problems. 

I will keep an eye on it for a few more days. 
Next week I can certainly also test the G-code, whether the milling machine does everything correctly. 

Then I would give you a conclusion and info for your overview. 

But already many, many thanks for your effort Tim.

Do you have Paypal or Patreon for a little support and love?

TimPaterson, 2020-12-13:
One way to help is to tune the delay, say to the nearest 0.1 second.  Try 0.3, then go to 0.2 or 0.4 depending on whether it worked.  You can add some margin when you're done, but it would be nice to know where the edge is.

BlackHawk3000, 2020-12-17:
Hello Tim,
I have now tested different file sizes and number of operations. 
Everything worked for me with a time of 0.2 :)

TimPaterson, 2020-12-17:
I have just released a new version that includes an option in a new "Advanced" section of the dialog to set the time delay.  The default is 0.2.  Being in the dialog not only makes it easy to change but can save a different value for each project, should some take longer that others.

My upload included adding linuxcnc.cps to the compatibility table.  However, I accidently left my default (Tormach) value in the tool change box, and I know that's wrong.  At the very least, it should have N10 in front to add a line number.  I would appreciate it if you could try it out with a tool change and let me know what works.  I will mark it "tested" when you give me the go-ahead,

closed 2021-05-31 by TimPaterson