TimPaterson/Fusion360-Batch-Post #11: Batch fails if operations are excluded by Suppress

LoganFraser, 2021-04-09:
I usually want to generate a breakpoint in my operations to add securing screws so I can remove the clamps. There doesn't seem to be an option to select just some of the operations to combine and restore rapids to, if I use suppress it just fails to generate a codefile. Is there any way this can be used for selected operations rather than all operations? 

[image]

TimPaterson, 2021-04-09:
Indeed, a setup with an operation that was suppressed failed.  Apparently this had never been tested, but it's fixed now.

When PostProcessAll was written, it was not possible for it to tell what was selected in the browser.  Now it can, but I can't say if or when I would go back and use it.  It's not quite as simple as it would seem, because if nothing is selected, that means do everything.

I also have needed a break between operations to add fixture screws.  I had inserted a manual operation to add the break, but now it ends up getting stripped out.  I haven't figured out how to prevent this, so I manually edit the G-code and add the break.

closed 2021-04-09 by TimPaterson

LoganFraser, 2021-04-10:
I think the easy solution to having a gap is just "duplicate setup" and have the before operations (drill/bore) in the first one and the rest in the second setup.