TimPaterson/Fusion360-Batch-Post #16: error when posting

gotdirt24, 2021-05-16:
i tried using it today im getting failed 0 files were written . 1 setups were skipped due to error; failed on setup (name of setup);
fusion360 reported an exception
funny thing is this file/setup has not changed last week it posted fine i have tried changing the timing and naming to test it 
 i did remove the preceded gode lines its was redundant

TimPaterson, 2021-05-16:
How many operations are in the setup?  If more than one, verify you can run the normal Fusion 360 post process command on each of them individually.  If that works, suppress all but one operation and see if some work with PostProcessAll and some don't.  You can use multiple select on the operations, right click and check/uncheck Suppress.  Make sure you have the latest version --  a bug in handling suppressed operations was fixed last month.

gotdirt24, 2021-05-22:
No matter what I do I keep getting that error 
I have suppressed each operation still have a error and I can post the regular way one by one 
Would you be willing to teamviewer in ?

Thanks mike

gotdirt24, 2021-05-22:
i tried installing post process all again with the newest github code it
does create a file to the output folder but empty 0kb

TimPaterson, 2021-05-23:
I forgot to ask the most basic question: What post processor are you using?

gotdirt24, 2021-05-23:
https://swissi2000.github.io/Fusion-360-Milling-Post-Processor-for-Centroid/

TimPaterson, 2021-05-23:
I was able to reproduce the problem with that post processor, which led to a simple fix. Get the latest version and give it a try (restart the add-in if Fusion 360 is running).

If it works for you, please also confirm what tool change code to use so I can update my compatibility table.

gotdirt24, 2021-05-23:
Very cool that’s maybe the most common post processor people are using for centroid it has a bunch of bells and whistles 
I will check it out tonight see how it goes

gotdirt24, 2021-05-23:
Added new code 
It’s doing something different

gotdirt24, 2021-05-30:
I have good news I did get it to work a few times and run the program everything worked as planned 
, there is a tool path grafting that is displayed Before the program starts that isn’t correct but the gcode is

TimPaterson, 2021-05-31:
I guess we're done then...

closed 2021-05-31 by TimPaterson