TimPaterson/Fusion360-Batch-Post #17: AttributeError: 'NoneType' object has no attribute 'groupdict'

satoer, 2021-05-30:
I made a simple bore path and when I try the script with the option "use individual operations" to check the "Restore rapid moves" (Why I want to use this script) I got the following error:

> 0 files were written. 1 Setups were skipped due to error:
Failed on setup Setup6: Fusion 360 reported an exception Traceback (most recent call last):
  File "C:/Users/Michel/AppData/Roaming/Autodesk/Autodesk Fusion 360/API/AddIns/PostProcessAll.py", line 973, in PostProcessSetup
    match = regBody.match(line).groupdict();
AttributeError: 'NoneType' object has no attribute 'groupdict'

[image]
Any idea what went wrong?

martinhansdk, 2021-05-30:
I'm seeing the same issue on a fresh install of the add-in.

TimPaterson, 2021-05-30:
What post processor are you using?

martinhansdk, 2021-05-31:
I'm using the GRBL post from https://github.com/martindb/mpcnc_posts_processor.

TimPaterson, 2021-05-31:
That post is incompatible with PostProcessAll (when using "individual operations") because it does not write the tool number G-code ("Txx") to the file. The GRBL post that comes with Fusion 360 works fine.

I did find the code in DIYCNC_Common.js which suppresses writing the tool change on line 421. You can edit this line to change this, leaving it as: 
`if (properties.toolChangeEnabled) {`

Once this change was made, the post outputs the tool number and works with an updated PostProcessAll. There was a bug in PostProcessall that did like an M-code on the line before the tool change, and that has been fixed. It also now reports the specific error if the tool change is not present.

satoer, 2021-06-01:
I'm using this post processor for the Onefinity cnc:
https://cam.autodesk.com/posts/post.php?name=onefinity
Is there something I should change also?

TimPaterson, 2021-06-02:
Looking at the Onefinity post I see a small block of code starting at line 561 that makes the tool command a comment if it's the first section.  It looks like this:
```
    if (!isFirstSection()) {
      writeBlock("T" + toolFormat.format(tool.number), conditional(getProperty("useM06"), mFormat.format(6)));
    } else {
      writeComment("T" + toolFormat.format(tool.number));
    }
```
Just delete all lines of this block except the second one, leaving:
```
    writeBlock("T" + toolFormat.format(tool.number), conditional(getProperty("useM06"), mFormat.format(6)));
```

satoer, 2021-06-02:
That seems to do the trick. Don't know if the rapids are actually restored (have not tried the GCode yet) But it finished the script without errors.

Thanks Tim.

closed 2021-06-02 by TimPaterson