TimPaterson/Fusion360-Batch-Post #19: Operation with disabled coolant gets deleted when combined

lutorm, 2021-07-05:
I've been using this plugin successfully with Personal Edition for a while, but recently I ran into an issue: I had several drill operations which used disabled coolant (M9). When these were combined, everything past the M9 was deleted. I assume that the M9 is used as part of some heuristic for when an operation is over?

I created a quick example. This is the result of the single-operation post:

```
(1001)
(TEST)
(T2  D=4.7 CR=0. TAPER=118DEG - ZMIN=-6. - DRILL)
G0 G90 G94 G17
G21
G53 Z0.

(DRILL2)
M5
M9
T2 M6
G53 Z0.
S6193
M3
G4 P3.
G54
M9
G0 X20.47 Y-1.557
Z15.
Z5.
Z4.
G1 Z-6. F437.
G0 Z5.
Z15.

M9
G53 Z0.
M30
```

But when combined with subsequent, similar, drill op, what results is this: 

```
(1 Setup1)
(T2  D=4.7 CR=0. TAPER=118DEG - ZMIN=-6. - DRILL)
(T3  D=23. CR=0. TAPER=118DEG - ZMIN=-6. - DRILL)
G0 G90 G94 G17
G21
G53 Z0.

(DRILL2)
M5
M9
T2 M6
G53 Z0.
S6193
M3
G4 P3.
G54

T3 M6
G53 Z0.
S1265
M3
G4 P3.
G54
M9
G0 X20.47 Y-1.557
Z15.
Z5.
Z4.
G1 Z-6. F437.
G0 Z5.
Z15.

M9
G53 Z0.
M30
```

Note that all that remains of the "drill2", with tool 2, is the preamble up to where it would assert the coolant is disabled, then it goes directly into the second drill with tool 3.

TimPaterson, 2021-07-06:
Yes, M9 is used as an end of operation marker.

I was unable to reproduce this using the tinyg.cps post that comes with Fusion 360.  When I turned off coolant, it did not add any new M9 commands, it just left off the M8 at the beginning.  Is that the post processor you're using?

lutorm, 2021-07-08:
I have a forked version with some changes. Maybe they've updated the one supplied with F360. Let me try porting my changes over.

TimPaterson, 2021-08-17:
The latest version allows you to specify exactly which G-codes mark the end of an operation. You can remove M9 from the list if they occur in the middle of the program.

closed 2021-08-17 by TimPaterson