TimPaterson/Fusion360-Batch-Post #27: Linuxcnc G53

alkabal, 2021-10-07:
Hi

First : many thanks for sharing this tool.

I get a error message trying to use your plugin with linuxcnc lathe post : 

> 0 files were written. 1 Setups were skipped due to error:
> Failed on setup Configuration4: Fusion 360 reported an exception: 3 : ReferenceError: G53 is not defined
> Error at line: 1

Can you help for that ?

Best regards

TimPaterson, 2021-10-08:
Now I realize this may be the same as Issue #24. I don't understand how Fusion 360 can throw an exception when using their API but it works when using their built-in command. There is info about how to debug it there, but there was no response on that and I'm not sure it would lead to a solution anyway.  I suspect a problem with the post processor itself or how it gets initialized.

A possible work-around would be for PostProcessAll to take the individual output files and stitch them together, but I'm not sure that's worth effort over just running them one at a time.

enricobono, 2021-10-20:
Hello. I have the same issue with Mach3mill.
Weird part is that I'm not using G53 (nor G28), so it's nowhere in the generated `.nc` files.

How can I help you to figure out what's wrong with F360?

closed 2021-12-11 by TimPaterson