TimPaterson/Fusion360-Batch-Post #33: Post only 1 operation

Hunteruga101, 2021-11-17:
Thanks for the great addon! he helped me a lot. True, I tried to add the ability to generate code only from the selected operation, but I could not. Can you tell me which part of the code is responsible for selecting input data? How do I make sure that the code is generated only from the selected operation, and not from the selected setup?

TimPaterson, 2021-11-17:
The add-in works by setups -- it does not look at selected operations. A setup represents a G-code file, so it generally doesn't make sense to only consider a single operation. But if need be, you can right click on the other operations in the setup and suppress them.

closed 2021-12-11 by TimPaterson

paramax55, 2021-12-28:
A single operation is actually what makes sense on the small 3018 CNC. It's more like a 3D printer with a spindle than anything else so cuts are long and slow. Tool changes are completely manual and probing has to be done after each tool change. Sometimes I even split up a single operation if it is going to take a while. I've had a single operation take more than 9 hours before.

TimPaterson, 2021-12-28:
I'm afraid you've completely missed my point. A setup corresponds to a file. If you only want one operation in a file, it should be the only operation in the setup. (You can select which setups you want PostProcessAll to generate.)

paramax55, 2021-12-28:
I may have missed your point and I may be doing it wrong, myself. But I just sort of went about this in the way that made sense to me. As an example, I have been whittling away at a part for the last couple of weeks. I made a setup for the bottom and I separated that into 15 different nc programs. Then I flipped it over and made a second setup for the top. I have 10 nc programs so far, but will probably end up with about 15 for that as well. It is a process that works well for this little machine since I do a lot of rest machining - nibbling away at the part a few hours at a time.

What drew me to PostProcessAll (and I believe the OP) was the idea of doing what I am already doing, but restore rapid movements. I right-clicked on [T1] Adaptive27(2), selected PostProcessAll (instead of Post Process) and was hoping to output a nc file on that operation. Of course you know that didn't work. What had happened was... the computer churned for a couple of hours, I clicked cancel and went to bed. When I got up in the morning, there was a message saying the process had failed.

It would be awesome to be able to select just one operation and fix the rapid movement problem with PostProcessAll.

Hunteruga101, 2022-01-12:
I don't see the difference between a small and a large machine. in any case, fast movements in adaptive machining speed up the process considerably. of course, when processing simple programs like a contour, there is not much difference.
so use the creator's ingenious program and enjoy :)
I work on a home-made, heavy, machine, made of polymer concrete, and this program saves me a lot, since there are few projects, and the price of work is too low to buy a subscription. although I would gladly pay for it, if it were possible.
nevertheless, from time to time, I try to find a solution to the issue, since I myself am studying python, but unfortunately, I don’t know what outgoing data fusion provides, and I don’t know how to get around with the issue of highlighting only one setting and not the entire setup.
p.s. thank you very much to the author for the program
pohto cnc below:
`https://sun9-82.userapi.com/impg/kafoQ8Gr-YMH_MrDLglwljEYLTYYSYQMNUuBAQ/GHXu8G-9LFk.jpg?size=900x1600&quality=95&sign=d2cbc08c0f435e7477bc5653a1725121&type=album`

paramax55, 2022-01-12:
That is a hefty looking machine, for sure. My last project had over 30 nc programs and it took 3 weeks to finish. I think I could have done it with less than 10 programs and finished in 1 day with a machine like yours. And that is the difference I am talking about when I compare a large machine to a small one.

I might be wrong, but I picture a workflow like this with a big machine: load your part into fusion, set the stock, make all of your toolpaths like they do in the youtube videos, then export your big, corrected .nc program. But that doesn't work with a little machine. Because it can take 3 weeks to make a little part, there is a lot of "run this nc program on this small portion and see how it goes." Then make another nc program based on your previous results, then repeat, repeat, repeat until your part is all cut out. That's the reason you would want to run the post on just one operation within a setup. And I haven't been able to figure out how to do that.

Hunteruga101, 2022-01-12:
can i see your machine? maybe in private messages? I started out with a small machine before creating a heavy machine. and the projects were complex. there was no such thing that the processing time differed by a couple of weeks. maybe the problem is not in the program, but in the approach to the project

paramax55, 2022-01-12:
If these pictures loaded correctly, one is a picture of the machine and the other is a picture of the part that took 3 weeks sitting on the beginning of another of the same part for the other side. It is a 3018 CNC. They are basically a beefed-up 3D printer with a spindle. I have settled on two tool sizes. I use a 3mm router bit for the bulk of my cutting and then use a 1mm for finer details. Tool load for the 3mm is .010" at a feed rate of 20"/minute and .010" depth of cut. The 1mm tool uses a 8"/minute feed rate, .008" tool load and .005" depth of cut. I spin both at 6,000 rpms. Any heavier cutting than that and you will get chatter. If you go too far, the chatter will be so severe that it will vibrate the machine across the table before it breaks the bit. And climb milling is absolutely NOT available for the reasons of chatter.

[image]
[image]

paramax55, 2022-01-14:
I used to have a machine with 3 spindles that were water cooled and spun at 24000 rpm. It was for routing out circuit boards. It was only intended for x-y movements and z plunging. it was made of steel and granite and weighed 6,000 pounds. It didn't chatter. But that was 20 years ago and those days are long gone.

This machine has been great for what it is and, as long as you stay within it's limitations, it will do a lot of cool things - eventually. The whole setup, including the cabinet, raspberry pi, keyboard, home made oiler, and everything was about $500.00. It has been well worth it for everything I have learned. My long term goal is to CNC a Precision Mathews machine. This GRBL controller can run one of those and they are less than $40.00. Then, as my needs increase, I think I'll change over to an Acorn controller.