TimPaterson/Fusion360-Batch-Post #38: "restore rapid moves" greyed out

paramax55, 2022-01-12:
I have been playing with post process all and I am having an issue with "restore rapid moves." I tried it on a previous part and the option wasn't greyed out. But I had too many operations going on by that time and my part was almost finished, so I gave up until now. I am only on the second operation of this part, so I am trying post process all. My first attempt generated a valid G-code file and that is good. But I need to restore the rapid moves and that option is greyed out. I don't know what types of things could cause this for me to even start looking for the problem.

TimPaterson, 2022-01-12:
The availability of `Restore rapid moves` and everything else in the `Personal Use` section is simply controlled by the `Use individual operations` option.  If `Use individual operations` is not checked, PostProcessAll does not examine the G-code at all, it just passes on what Fusion 360 generated.

paramax55, 2022-01-12:
That was it. Thank you.

closed 2022-01-12 by paramax55