TimPaterson/Fusion360-Batch-Post #41: issue with heidenhain post processor tool changes.

dbrown1225, 2022-06-02:
I've been teaching myself fusion360 for about 2 weeks now, I've loved this add-in! Up until this point I've been only using the Fanuc post processor which hasn't had any issues, however, i would like to attempt to use fusion for my HermleC50U with Heidenhain ITNC530 controls. I got the Heidenhain post processor, which has a ".h" output file type, and If I try to use the regular "post process" for my 1 tool it will work fine, however when going to use "post process all" I get this error

Failed on setup Setup5: Tool change G-code (Txx) not found; this post processor is not compatible with Post Process All.
Failed on setup Setup6: Tool change G-code (Txx) not found; this post processor is not compatible with Post Process All.

I'm not too familiar with editing post processors, so I'm not sure if there is something I can change to help this. Thanks in advance!

TimPaterson, 2022-06-02:
To be honest, my first thought is that someone who has a 5-axis machine ought to be able to pay for a subscription to F360. Then you would use PostProcessAll without selecting "use individual operations" and your problems should go away.

After a quick look at the post processor, I didn't recognize anything looking like G-code at all.  A glance at the output could easily confirm this.  (The "individual operations" options works by parsing G-code.)

TimPaterson, 2022-06-02:
You will need a separate setup (and thus a separate output file) for each tool. You can run them one by one or edit them manually to splice them together with tool changes if you know the control format.

dbrown1225, 2022-06-02:
Okay gotcha, thank you for your help!

closed 2022-06-02 by TimPaterson