TimPaterson/Fusion360-Batch-Post #42: Not compatible with GRBL post processor?

water-rhatt, 2022-07-14:
I am trying to use the macro and am getting this error:

```
0 files were written. 1 Setups were skipped due to error:
Failed on setup Underside: Tool change G-code (Txx) not found; this post processor is not compatible with Post Process All.
```

Using the GRBL.cps from Autodesk.

 If I just post just using Fusion and selecting a single tool, I get the TX M6 output. 

For example, here is a random file I just generated without using PostProcessAll:

1002_1_T12.nc:
```
(1002)
G90 G94
G17
G20
(Drill1)
T12 M6
S5000 M3
G54
G0 X5.497 Y0
Z0.6
Z0.2
G1 Z-0.5 F40
G0 Z0.2
X15.997 Y0.0096
G1 Z-0.5 F40
G0 Z0.2
Z0.6
G28 G91 Z0
G90
G28 G91 X0 Y0
G90
M5
M30
```

Am I doing something wrong? Is there a grbl post-processor that is compatible?

TimPaterson, 2022-07-14:
I made sure I was using the latest post (dated yesterday!) and didn't have a problem -- unless the "output tool number" property in the NC Programs dialog was not checked. The options used by PostProcessAll should be whatever options were used by the last manual post.

You can look at the output for the individual operation as PostProcessAll sees it when this fails. A temporary file is left behind; on my machine it's `C:\Users\Tim\AppData\Local\Temp\8910.nc` (so put your user name in for "Tim"). It should be identical to what you get with a single manual operation.

water-rhatt, 2022-07-15:
As mysteriously as it didn't work yesterday, it worked today. I'm closing the issue. Hopefully, it won't reappear.