TimPaterson/Fusion360-Batch-Post #43: Post Process All appears to be broken now

GreyfoxTime, 2022-08-05:
Trying to output a process with a few tool changes, the processor fails and says it can't find the C:\Users\appdata\local\temp/8910.nc
was working last month or so.

TimPaterson, 2022-08-05:
It does seem there was a Fusion 360 update yesterday. However, I did a quick test on a sample project and it worked fine.

I would need the exact text of the error message (so I can search for it in the sourece code) to be sure where the error is occuring.

GreyfoxTime, 2022-08-05:
0 files were written. 1 Setups were skipped due to error:
Failed on setup Setup2 (4): Unable to open C:\Users\kinta\AppData\Local\Temp/8910.nc

here's the full error, working off a document with multiple setups, currently have only one selected to process tho. doing everything as I normally do so not sure what changed here.

GreyfoxTime, 2022-08-05:
tried with a fresh document and still getting the same issue, have tried uninstalling the post processor and adding it again as well.

TimPaterson, 2022-08-05:
I rried a larger project (20 setups, many tool changes) and didn't have a problem. This is probably specific to your post processor - which one are you using? Are you using your own fixed copy or is it updating with Fusion 360?

The error means that PostProcessAll asked Fusion 360 to post process one operation in a setup and it never finished and didn't report an error. You can increase the time PostProcessAll waits for F360 to finish under the Advanced option in the PostProcessAll dialog. Start by increasing the "Initial time allowance" quite a bit, say to 1 second. This will slow things down considerably on larger projects, as PostProcessAll won't check for completion until the full time completes. It doubles the time for each retry. Retries haven't seemed to be as effective as longer initial time.

GreyfoxTime, 2022-08-05:
I tried the latest centroid post from fusion along with my previous edited post I was using which worked just fine previously, both are causing the same issue tho. I had tried the 1 second timing before posting as well and it didn't help. The processor will ask to save to a location, which I do, then it does the same request again, the nc file does save but it's only ever the first operation and it repeats the request to save 3 times before stopping.

TimPaterson, 2022-08-06:
Wait - what's this "processor will ask to save to a location"?  I just used the current Autodesk-provided Centroid post and it ran fine on a small project. There aren't any prompts after clicking OK on the PostProcessAll dialog.

GreyfoxTime, 2022-08-06:
I'll try and get some pictures and what not here when I have a bit more time today. Thank you for helping me look into this tho as it is much appreciated. When I click the button to process the operations, it opens a secondary window asking where I'd like to save the file to and what name to give it tho. I have the fusion360 processor setup to open a second window asking for the save location and name as well.

GreyfoxTime, 2022-08-06:
one thing to note is it's always wanting to save with the default name 8910.nc, I never put that in but it always defaults to it since I've been having this issue. Not sure if that helps at all tho.

GreyfoxTime, 2022-08-06:
seems I found the issue!, in preferences under manufacturing the "show the save as dialog when posting from nc programs" needs to be left unchecked. I do like the functionality of the save as prompt tho as it helps with organizing lots of files for production runs more easily.

TimPaterson, 2022-08-06:
PostProcessAll creates a temporary file (named 8910.nc) for each individual opertion, then copies the content into the main file in the folder specified in the PostProcessAll dialog. It can't possibly work if you're changing the file name behind its back. It makes no sense at all for you to have it prompt for a file name when the output location is set in the PostProcessAll dialog -- just set it there.

closed 2022-08-06 by TimPaterson