TimPaterson/Fusion360-Batch-Post #44: Request: Skip first tool change option

Billiam, 2022-08-21:
It would be helpful to have an option to skip this first tool change when post processing a setup with multiple tools.

My machine doesn't have direct support for tool selection, but my sender (cncjs) will see an M6, issue a pause, and allow a manual tool change.

It doesn't have any way to know that the first requested tool is already loaded, so it can't intelligently skip it.

When Fusion supported tool changes on their free plan, I edited my post processor to skip this first tool change, but that isn't currently possible when using Post Process All.

TimPaterson, 2022-08-21:
It's a lot of work to add options, but for someone who can edit their post processor this should be an easy hard-coded change.  I think you need to combine the conditionals on lines 1049 and 1051 into one, so it becomes
`if not fFirst and toolCur != toolLast:`
This will cause the code that skips a tool change at lines 1061 - 1063 to be executed on the first op.  Then you need to add
`toolLast = toolCur`
before the `continue` to set `toolLast` on the first op.

I haven't tested this, so no guarantee it's correct.

Billiam, 2022-08-22:
Thanks, I haven't had any luck there so far, but I'll play around with it a bit more

closed 2022-08-22 by Billiam