TimPaterson/Fusion360-Batch-Post #45: Rapid Restoration: If first non-rapid move is a non-xy arc, feed rate will not be added causing errors

Billiam, 2022-08-22:
I had a facing operation using an Z/X arc lead in, that creates the following gcode by default:

```gcode
(T2  D=6.363 CR=0 - ZMIN=-13 - flat end mill)
G90 G94
G17
G21
(When using Fusion 360 for Personal Use, the feedrate of)
(rapid moves is reduced to match the feedrate of cutting)
(moves, which can increase machining time. Unrestricted rapid)
(moves are available with a Fusion 360 Subscription.)

(Face12)
S18000 M3
G54
G0 X67.474 Y-11.329
Z15
G1 Z6.1 F2700
G18 G3 X60.974 Z-0.4 I-6.5 K0
G1 X60.692
```

When using PostProcessAll, the feed rate is restored for the G1 on the last line, but not for the arc above:

```gcode
G90 G94
G17
G21
(When using Fusion 360 for Personal Use, the feedrate of)
(rapid moves is reduced to match the feedrate of cutting)
(moves, which can increase machining time. Unrestricted rapid)
(moves are available with a Fusion 360 Subscription.)

(Face12)
T2 M6 (Downcut flat 1/4" plus runout)
S18000 M3
G54
G0 X67.474 Y-11.329
Z15
G00 Z6.1 (Changed from: "G1 Z6.1 F2700")
G18 G3 X60.974 Z-0.4 I-6.5 K0
G1 X60.692 F2700.0 (Feed rate added)
```

It seems like arcs are being accounted for in the MotionGcodeSet, but I'm not clear on whether it's accounting for the `G18`, or just treating the whole line as non-movement.

TimPaterson, 2022-08-23:
There were a few bugs/limitations in the G-code parsing that led to this problem. I have committed a bug fix which works on your provided test case, so give it a go.

TimPaterson, 2022-08-23:
After further review, I think my fix introduces a new bug, so hold on for a bit...

TimPaterson, 2022-08-23:
It should be fixed now.

Billiam, 2022-08-23:
Thanks, looks good to me :+1:

closed 2022-08-23 by Billiam