TimPaterson/Fusion360-Batch-Post #49: Restore rapids modifying g28 strangely

jeremycranfill, 2022-11-23:
I am using the grblhal post processor and g28 safe retracts. Using restore rapids seems to work fine except for at the end

`G00 Z51.987 (Changed from: "Z51.987")
G01 Z0.0 F2512.0 (Changed from: "G28 G91 Z0")
G90
G28 G91 X0 Y0
G90
M5
M30`

The g28 g91 zo change seems off, and the following G28 G91 X0 Y0 causes a crash. Not sure what could be causing this

TimPaterson, 2022-11-23:
Could you attach a complete G-code file?  The smallest test case that demonstrates the problem would be best. Also give me a link to the exact post processor you're using (or attach it as well).

jeremycranfill, 2022-11-26:
[https://pastebin.com/9FiMCebh](url)heres a sample gcode.  https://github.com/raenji-sk/grblHAL_Fusion360_Post_Processor heres post processor

TimPaterson, 2022-11-26:
I was able to reproduce this issue and have committed a fix. PostProcessAll now recogizes homing G-codes (G28/G30) as special and doesn't alter them. Give it a try and let me know.

jeremycranfill, 2022-11-27:
Looks fixed to me. Thanks a bunch

closed 2022-11-27 by TimPaterson