TimPaterson/Fusion360-Batch-Post #50: Unable to open output file

sharab85, 2022-12-03:
Hello, 

First of all I would like to thank you for making this very useful tool! 

I have been using it for awhile with my Milltonic's mill and the milltronics.cps post.

I got a new lathe that uses a siemens 808D controller, the siemens turning post that fusion has seems to be working with my machine but when I try and use the post process all utility I get the following error. 

  Unable to open output file. Found the file with extension '.failed' instead of ''. Make sure you have the correct file extension set in the Post Process All dialog.

Followed by 

0 files were written. 1 Setups were skipped due to error:
Failed on setup Setup1: Unable to open C:\Users\shara\AppData\Local\Temp/8910

Any help with this would be greatly appreciated.

TimPaterson, 2022-12-03:
Since the "instead of" sentence apparently ends in a blank, maybe (as the message says) you haven't set the "Output file extension" field in the PostProcessAll dialog?  For this post it should be ".mpf".  A very simple example worked for me.

sharab85, 2022-12-04:
That fixed it I still had it with the output file type for my mill. 

Than you!