TimPaterson/Fusion360-Batch-Post #51: Problem with missing "M6" when using grblHAL_Fusion360_Post_Processor

R-Ronald, 2023-01-28:
Using the grbl post processor that is in the Fusion 360 cloud library, PostProcessAll produces an output file with "M6" tool change command in the same line as the tool number.

Using the experimental grblHAL_Fusion360_Post_Processor from [https://github.com/raenji-sk/grblHAL_Fusion360_Post_Processor](url), the "M6" command is missing from all lines with the tool number in the PostProcessAll output file.

But, when the experimental grblHAL post-processor is run directly against a single operation, the "M6" command **is** included in the output file.

I have just posted this problem on the PrintNC site (in cad-cam-design section) and have received a replay that indicates that someone else has noticed this problem.

A key difference between the two post-processors is that one is located in the Fusion 360 cloud library. The experimental one is not available in their library, so it was stored locally. I'm not sure if there's any specific requirements by Fusion or by PostProcessAll for the location of the post processor .cps file.

LoganFraser, 2023-01-29:
I can confirm the same issue on my machine.  Post Process is including G6 T# but Post Process All is only including T# as shown: 
[image]

TimPaterson, 2023-01-30:
Normally you could run the built-in Post Process command once to set various option (such as checking the "Output M6" checkbox), and then those options will be used when PostProcessAll is run. However, this does not seem to be happening. This may be a change to Fusion 360, as I found it was now happening with other post processors also. I have put a message on their support forum to see I can find out more.

In the meantime, the work-around is to edit the post processor file (.cps), and change the default value of the property. For this particular post processor, the value for M6 is on line 122, which would be changed from "false" to "true" (but no quotes). If you look at the code above and below, you will see that all the properties in the dialog are listed there and you can change any of them if you like.

LoganFraser, 2023-01-31:
I think it's possible you've misread the description of what line 122 does: 
[image]
Setting it to True DISABLES posting M6 commands. We don't want that. :D

R-Ronald, 2023-01-31:
Thanks for looking into this so quickly. And so well. I can also confirm success with the workaround.

TimPaterson, 2023-01-31:
That's great. I'm leaving this issue open while hoping to get action or a response from Autodesk.

R-Ronald, 2023-01-31:
> image

It's early and the text is bizarrely written using a _double 
negative_.....but TRUE works in the file when FALSE does not.

Our goal is to "allow the output of M6 on tool changes". So we do not 
want  want to "disallow" that by setting "Disable to disallow". Instead, 
we want to enable (using TRUE) to allow.

RR

LoganFraser, 2023-01-31:
Agreed, the phrasing is confusing.  My initial thought was setting true would disable M6 outputs.

TimPaterson, 2023-02-03:
Here is the conversation with F360 support:
https://forums.autodesk.com/t5/fusion-360-manufacture/using-api-to-post-process-always-using-default-post-properties/td-p/11716679
First they said the change was intended, then they said there had been no change.  When I assured them it was changed, they went silent.

R-Ronald, 2023-02-06:
One more update.
I was pretty surprised today when my 10mm BAP tool tried to drill a 2.5mm hole. Didn't work too well. 

My fault for not checking for M6 in the file. I guess I actually did check by running the program, but not in a good way. :-)

I had a grblHAL.cps file with "True" in line 122 and it was still not working. I could generate bad files at will. However, I was able to fix this by closing F360 and reopening. Seems like updating the line 122 text in the file needs to be done **before** Fusion is running. It doesn't pick it up if you change it while it is running.

RR

leomgrab, 2023-09-11:
I am having this issue, i am using fusion360 grbl post, (with out without "true" on line 122)
 also reseting my system, no change.
the first tool T1 has the m6 
(Engrave1)
T1 M6
S9000 M3
G17 G90 G94
G54
G0 X68.499 Y-0.061

when i look at the code for the second tool, there is no m6 command 
(2D Pocket1)
S9000 M3
G17 G90 G94
G54
G0 X4.771 Y-57.038
Z16.73
maybe you could help guied me so i dont have to edit every program after post?

TimPaterson, 2023-09-11:
You mention no M6, but the code doesn't show the `Tx` command either. I couldn't repro a problem using the latest post (44083 dated a few weeks ago), but I did need to edit it to turn on the "Output M6" option.

leomgrab, 2023-09-11:
@TimPaterson  Thank you for you'r speedy responce! :) 
I have figured out the M6 issue i was having!   
                     (using a tool dir from online)
                     all the tools are labled (T1) it could not seprate the tools in the GRBL-post because all tools were labled (T1) 

                    when you said "but the code doesn't show the Tx command either"
                    can you explaine this, is the TX needed?

(Engrave1)
T1 M6
S9000 M3
G17 G90 G94

(2D Pocket1)
T2 M6
S9000 M3
G17 G90 G94
G54

[image]
[image]

TimPaterson, 2023-09-11:
Yes, if the tool number is the same it doesn't output any tool change commands.

TimPaterson, 2025-07-18:
`PostProcessAll` now uses `NC Programs`, which provides access to all post processor options. Please confirm that v2.0 resolves the original issue.

leomgrab, 2025-09-26:
> Assuming this is fixed.

yes thank you!