TimPaterson/Fusion360-Batch-Post #52: Identical tool number assigned to two different tools-creates problem for tool change

R-Ronald, 2023-03-12:
I created a simple model in Fusion and build two tool paths.

1. A facing operation with a 9.6mm diameter indexed end mill
2. A 2d adaptive clearing operation to partially mill some slots

When I run PostProcessAll, I get gcode with both tools as "M1":

(1 Setup1)
(T1  D=9.6 CR=0 - ZMIN=-4.05 - flat end mill)
(T1  D=4 CR=0 - ZMIN=-13.05 - flat end mill)
<snip:

(Face1)
N35 T1 M6
<snip>

There are no other tool change commands in the file. Thus, the 2d-adative operation will try and use the wrong tool.

For a workaround, I believe that I can just insert a "M6 T2" in the line after the (2D Adaptive2) comment  that marks the start of the 2nd operation (to force the program to pause for a tool change).

R-Ronald, 2023-03-14:
> What post processor are you using? Do you have "Use individual operations" checked in PostProcessAll? What does the output from the built-in post processing look like?

Post Processor is GrblHAL.

"Use Individual Operations" is checked.

The beginning of the two individually post-processed files (using GrblHAL) are:

#1

    (FaceOp)
    (T1  D=9.6 CR=0 - ZMIN=0 - flat end mill)
    N10 G90 G94
    N11 G17
    N12 G21
    (When using Fusion 360 <snip> a Fusion 360 Subscription.)
    N13 G28 G91 Z0
    N14 G90

    (Face1)
    N15 T1 M6
    N16 S18000 M3
    N17 G4 P2
    N18 G54

and #2

    (2dadaptive)
    (T1  D=4 CR=0 - ZMIN=-8 - flat end mill)
    N10 G90 G94
    N11 G17
    N12 G21
    (When using Fusion <snip>  with a Fusion 360 Subscription.)
    N13 G28 G91 Z0
    N14 G90

    (2D Adaptive2)
    N15 T1 M6
    N16 S24000 M3
    N17 G4 P2

Using the default Grbl/grbl post procesor, I get:

#1

    (test3)
    (T1 D=9.6 CR=0 - ZMIN=0 - flat end mill)
    G90 G94
    G17
    G21

and #2

    (test2)
    (T1 D=4 CR=0 - ZMIN=-8 - flat end mill)
    G90 G94
    G17
    G21

TimPaterson, 2023-03-14:
Clearly PostProcessAll is working perfectly -- you've asked for tool #1 for both operations and that's what you're getting.  If you want to change tools, you must give them different tool numbers in your tool library.  If you had a paid subscription to F360 and didn't need to check the "Use individual operations", you'd get exactly the same thing.

R-Ronald, 2023-03-15:
Thanks. I should have realized that the user controls the tool number. Until I saw your second sentence, I thought that the post-processors assigned a number. But it's obvious now. I'll go thru my tools and change their numbers.

closed 2023-03-15 by R-Ronald