TimPaterson/Fusion360-Batch-Post #55: Post Process All Fail

confused1234, 2023-05-22:
I have no idea what is going on, it worked for a very long time now all of a sudden after a Fusion360 update I can no longer use the process all. I keep getting this fault. any recommendations?
[image]

TimPaterson, 2023-05-22:
Believe the error message.  

What post processor are you using? This is probably a variation of Issue #51. Your post may have an option to turn on/off tool change commands in some circumstances, and the default is off. Or it could be the post was changed by the update to no longer output the tool change.

relentlessly, 2023-05-22:
I am also having this issue.
[image]

Tried a whole bunch of combinations of settings in the default/stock postprocess window and the plugin "post process all" window with no luck
Using the linxucnc post processor:
`AppData/Local/Autodesk/Autodesk Fusion 360/CAM/cache/posts/linuxcnc.cps` (EMC2)

It appears something fusion 360 did completely broke the plugin (or probably just stopped emitting tool change gcode)
I looked through `linuxcnc.cps` but it is a little too cryptic to me
Any advice or help would be appreciated

TimPaterson, 2023-05-22:
I tried using the latest LinuxCNC post (revision 44066) and was unable to reproduce the problem. Run the built-in post process command on a single operation and see if the .ngc output includes a tool change command ("T") near the start.

relentlessly, 2023-05-22:
I ran the build in post process on a single operation as suggested. 
There is not Tool change command
```
%
(FACE-SINGLEOP)
(T9 D=6. CR=0. - ZMIN=-1.4 - FLAT END MILL)
N10 G90 G94 G17 G91.1
N15 G21
(WHEN USING FUSION 360 FOR PERSONAL USE, THE FEEDRATE OF)
(RAPID MOVES IS REDUCED TO MATCH THE FEEDRATE OF CUTTING)
(MOVES, WHICH CAN INCREASE MACHINING TIME. UNRESTRICTED RAPID)
(MOVES ARE AVAILABLE WITH A FUSION 360 SUBSCRIPTION.)
N20 G53 G0 Z0.
(FACE)
N25 M0
(MANUAL TOOL CHANGE TO T9)
N30 S5000 M3
N35 G17 G90 G94
N40 G54
N45 G64 P0.01 Q0.01
N50 G0 X99.4 Y1.137
N55 G43 Z15. H9
N60 G1 Z5. F495.
N65 Z-0.1 F333.33
N70 G18 G3 X98.8 Z-0.7 I-0.6 K0. F495.
N75 G1 X98.51
N80 X-3.01
N85 G17 G2 Y5.232 I0. J2.048
N90 G1 X98.51
N95 G3 Y9.327 I0. J2.048
```

[image]

confused1234, 2023-05-22:
You guys nailed it, I use Centriod and once I deselected the Manual tool change it started working again. I was really getting bummed because I thought Autodesk wised up and turned that off. Now Im debating if I want to add a 4th axis, $1600 add on for one year is stiff for a hobbyist. 

You guys rock!

closed 2023-05-22 by TimPaterson