TimPaterson/Fusion360-Batch-Post #56: Error. Cannot locate the simple error log generated from the Post Processing.

AndrewZAP1977, 2023-06-07:
Hello!
Sorry about my English!
Thanks for a nice addon.

Could you explain what is this error?
It does not appear in all models. In some models addon works great. And in other models this error appears and the g-code file is not uploaded.
If it's important - I use the mach3mill postprocessor.
I tried to change different settings in the addon window and in the Settings tab, but nothing helps!

With all respect, Andrew.
[image]

TimPaterson, 2023-06-07:
Are you using the free ("Personal Use") version? Do you have the "Use individual operations" box checked? 
 - If not, does the built-in post proccess command work for the whole setup?  
 - If so, does the built-in post process command work for each of the operations in the setup?

What are the operations in the setup?

AndrewZAP1977, 2023-06-07:
Thanks for the very fast response!
I have a version for personal use.
If I do not check the "Use individual operations" box, then the situation that I described in my question occurs.
When I check the "Use individual operations" checkbox, the addon normally unloads the g-code for the entire Setup.
I'm sorry, maybe I just didn't read the description of the addon very carefully, and it's supposed to work like that! And the error arose precisely because of the unchecked box?

With all respect, Andrew.

AndrewZAP1977, 2023-06-07:
Yes. When I checked the box, the addon worked fine with all models.

Tim, can I ask you one more question?
If I understood correctly from the description of the addon, it does not know how to unload the g-code for each operation separately.
Does it unload g-code in full for the whole Setup?
Or was I again not attentive and missed something in the description of the addon?

With all respect, Andrew.

TimPaterson, 2023-06-07:
Sorry, I do not understand your question. But here is the explanation of the "individual operations" checkbox, copied from the main page:

> In the Personal Use section of the Post Process All dialog, you will find the option to break each setup into individual operations. This is required if you have the free version of Fusion 360 and there is a tool change in any setup. Post Process All will combine the G-code for each operation of a setup back into a single file for that setup

AndrewZAP1977, 2023-06-07:
Yes. Thank you.
I saw this explanation for the personal version.
I didn't formulate the question correctly. Sorry!!!
I meant, can the addon save each operation in a separate file?
At me saves sequence of operations in one big file.

With all respect, Andrew.

TimPaterson, 2023-06-07:
There is always one file per setup. The file name comes from the setup name. If you want a separate file for an operation, it must be in its own setup. You can easily break up a setup by duplicating it for each operation and then deleting all the other operations.

closed 2023-06-08 by TimPaterson