TimPaterson/Fusion360-Batch-Post #57: 2d Adaptive Clearing with multiple depths G1 in the Z direction instead of G0

bhowiebkr, 2023-06-16:
I’m trying to get a 2d Adaptive Clearing with multiple depths working. When the tool finishes a layer and G0 rapid over to start the next layer, it will traverse in the Z direction at G1 feed speed instead of a G0 move. 

See attached ngc (renamed to txt for github) files:

[1001.txt](https://github.com/TimPaterson/Fusion360-Batch-Post/files/11765758/1001.txt)
[Setup1.txt](https://github.com/TimPaterson/Fusion360-Batch-Post/files/11765759/Setup1.txt)

TimPaterson, 2023-06-16:
Yes, after determining the cutting height, all moves at or below that height are kept G1.  When I mill a thread, it plunges G0 to the bottom of the hole on the first pass before any cutting.  Then it adjusts the cutting height to the top of the thread as it cuts it, and on the second pass moves G1 to the bottom.

If you thought it was worth the effort, it might work to use separate operations for each cut height.  But it will home between operations which might eliminate any gains.

bhowiebkr, 2023-06-20:
Ok that makes sense, I guess you have no way of knowing if the movement back to the cutting height is outside of the material which would be fine as a rapid move or if it’s plunging into material in which case you want to go at feed speed.