TimPaterson/Fusion360-Batch-Post #59: Failed to post: Masso.cps Onfinity Foreman

DeadspaceDan, 2023-09-18:
Error:
0 files were written. 1 Setups were skipped due to error:
Failed on setup Split Enclosure 500: Fusion 360 reported an exception: 3 : Cannot locate the simple error log generated from the Post Processing.

Using Masso V3 Touch and their provided F360 Post: Masso.cps

TimPaterson, 2023-09-19:
I ran a quick test using masso.cps on a small project and it worked OK.  Make sure you can post with the built-in F360 command -- it will give you better error reporting if there's a problem.

DeadspaceDan, 2023-09-19:
I am not sure what you mean. I have generated cut files from F360 just fine, but when i try pressing the Process All button the above error comes up and no output. I tried Processing Individual files, and several other options in Advanced but it all triggers the same error message. I am on a Mac if that helps as reference.

DeadspaceDan, 2023-09-19:
/facepalm  I thought i had tried it both ways. However it worked fine with i attempted it again with "Use individual operations" selected .. Well thank you for this great piece of software!

closed 2023-09-19 by DeadspaceDan