TimPaterson/Fusion360-Batch-Post #60: Fusion 360 reported an exception: 3 : Cannot locate the simple error log generated from the Post Processing.

alanbeswick47, 2023-11-21:
I have a Fusion 360 personal model, two tools using tried and tested ATC. This model has been working faultlessly for months.

Other models work perfectly today.

This one model gives this error

> 1 files were written. 1 Setups were skipped due to error:
> Failed on setup dolphin side 1: Fusion 360 reported an exception: 3 : Cannot locate the simple error log generated from the Post Processing.

Both setups will generate NC code using the standard Fusion 360 post processor. Mac has been rebooted - no improvement.

Same fault today on my laptop.

Any thoughts will be gratefully received
Thanks
Alan
Brisbane, Australia

TimPaterson, 2023-11-21:
The few times I've seen a failure with F360 reporting an exception, it also failed using the standard post command -- and gave an error message that told me what to fix.  Without that, I don't have a suggestion at this moment.

alanbeswick47, 2023-11-21:
Hi Tim
Many thanks for the prompt reply. I love your processor.

I have attached the .f3d file if you wish to chase the bug further. I’m somewhat python literate if you want some extra digging at my end.

Cheers
Alan

￼

TimPaterson, 2023-11-21:
I don't see an attachment.  GitHub has restrictions on attachments, so you may need to zip it first.

alanbeswick47, 2023-11-22:
Hi Tim
The f3d file left me OK :-(

I zipped it to change the suffix, that might help. If not, I’ll have to try some sort of file sharing service. Always a challenge!!

I wish this was easier.

Cheers
Alan
￼

dragoonmc, 2023-11-22:
I had mine doing this and it was because I have a personal license but had not selected the use individual operations

TimPaterson, 2023-11-22:
Great catch dragoonmc!  This is the same as Issue #59 reported 2 months ago.  I was smarter then, when I said:
> On closer inspection, I noticed the error message does not include the operation it was on when it failed. This should only be able to happen when the `Use individual operations` option (under `Personal Use`) is not checked.

Maybe the `Fusion 360 for Personal Use` section of the ReadMe is not prominent enough: 
> In the Personal Use section of the `Post Process All` dialog, you will find the option to break each setup into individual operations. This is required if you have the free version of Fusion 360 and there is a tool change in any setup.

alanbeswick47, 2023-11-22:
Thanks a lot for the help. I didn't realise the significance of the phrase "Use individual operations"

closed 2023-11-22 by TimPaterson

alanbeswick47, 2024-09-04:
Hi Tim
Great product, I love it and prefer it to the standard Fusion post processor. I recommend it to all my students.

Suggestion (to fix a minor irritation)
I use Mac at home and a shared Windows box attached to the CNC machine, both accessing a shared google drive. The path names to output and the post directories are different. Having to use C: is just sooo last millennium.

If there were a pair of radio buttons, say next to save defaults checkbox, that toggled between the Windows and the linux version of the directory path name strings, I would be an even greater fan of the batch command :-)

Eg
/Users/YourName/Documents
versus 
 C:\Users\YourName\Documents

Cheers
Alan
Brisbane
Australia

TimPaterson, 2024-09-05:
For your particular use case, it seems pretty easy to just add or remove the "C:".

I do see value in being able to keep more than one file/folder name (for both output folder and post processor), especially for use on different machines or in different configurations. Not likely to work on this any time soon though.

alanbeswick47, 2024-09-05:
Thanks. I thought it would be unlikely to get a fix for this. It falls under the great Australian saying, “Not a BBQ stopper”.

Cheers
Alan