TimPaterson/Fusion360-Batch-Post #67: 'Preserve order' in Pattern not respected

teckwee, 2024-02-16:
I used Patterns quite extensively recently and notice that the 'Preserve order' has different behavior in some designs. 

Essentially, the output for 'Preserve order' and 'Order by operation' is the same even though the former is clearly what I want when I simulated the operation. 

[image]

Happy to provide files and info if required. Thank you so much.

teckwee, 2024-02-16:
Sorry I just saw this... should have searched before - https://github.com/TimPaterson/Fusion360-Batch-Post/discussions/36

Is anyone aware of any updates? I'm pretty sure it works in one of our design.

TimPaterson, 2024-02-18:
There is no new information on this.

closed 2024-03-07 by TimPaterson