TimPaterson/Fusion360-Batch-Post #68: Restore Rapids adds transforms G0 that are already produced by the post processor into G00

meeloo, 2024-03-07:
I use a custom post processor (https://github.com/meeloo/Duet3Config/blob/main/Fusion%20360%20Post%20processor/duetcnc.cps) that already produces some G0 commands and it seems that when using the batch processor those G0 commands are transformed into G00 commands that my CNC controller (Duet3, RepRapFirmware based) doesn't know how to interpret and stops processing the file.

TimPaterson, 2024-03-07:
I'd be interested in seeing the G-code you're talking about. PostProcessAll should never modify an existing G0 command, and any line it does modify will have a comment showing the orignal line.

In any case, you can easily modify it to output G0 instead of G00 by editing PostProcessAll.py.  Find the strings with "G00" in it (lines 50 & 51) and change them to "G0".

meeloo, 2024-03-29:
sorry for getting back so late. indeed I was wrong and the changes you mentioned do fix the issue on my controller. Do you plan to make this change in the source? Is there a reason why G00 is preferred over G0?

closed 2025-01-28 by TimPaterson