TimPaterson/Fusion360-Batch-Post #69: Unable to get ATC to change

Turk1509, 2024-05-28:
Hi, when the first tool is finishes, instead of automatically changing, the spindle first retracts and then slowly feeds until it is to crash into the bed. This is on the Makera Carvera, 0.9.7 firmware using fusion 2.0.19215 x86_64
Any recommendations? Thanks

TimPaterson, 2024-05-28:
For this to be related to PostProcessAll, I guess you must have Fusion for Personal Use and have the "Use individual operations" box checked in the PostProcessAll dialog.  There's a box below that where you enter whatever G-code is needed for a tool change, so you need to make sure that's right.

Turk1509, 2024-05-29:
yep, i have the personal use and i clicked the use individual operations,
Should i  unclick the use individual operations? as currently i only checked it as i get this error message

"0 files were written. 1 Setups were skipped due to error:
Failed on setup ATC Test 3: Fusion 360 reported an exception: 3 : Cannot locate the simple error log generated from the Post Processing."

TimPaterson, 2024-05-29:
Now I'm confused about whether your problem is incorrect G-code or no G-code at all because of the exception.

The exception you're seeing is typically caused by using a setup name that is not a valid file name. PostProcessAll uses the setup name as the name for the G-code file, so it must be valid for your operating system (you can't use a quote mark [as in inches] for example).

Turk1509, 2024-05-31:
you were on the money, i had the default gcode but it was incorrect for my machine, this is with "Use individual operations" box checked
I changed the g code in the box as you said and rechecked "Use individual operations" and it seems to 
thanks for this amazing add on

closed 2024-05-31 by Turk1509