TimPaterson/Fusion360-Batch-Post #70: Skipping one setup

Kjellekasper, 2024-09-05:
I,m running the free Fusion 360. 
When running PostProcessAll it skips at least the last setup. 
Example:
When running 3 setups where all setups are using different tools, the last setup is skipped
When running 3 setups where setup 1 and 3 is using the same tool, setup 2 and 3 are skipped.
When running 3 setups using the same tool, all setups are skipped.
I normally use UCCNC.cps but have also tried Mach3Mill.cps. Both with the same result
When I do not tick "Use Individual Operations" all setups are generated without errors.

BR
Jan

TimPaterson, 2024-09-07:
I tested again with Mach3Mill and for the first time with UCCNC and they both worked fine for me.  When you say it skips a setup, is the file silently missing or does it report something?

Kjellekasper, 2024-09-07:
Hi Tim

I do think I did a rookie mistake, I have several different items in a
multibody file on wich I only have a single operation but with different
tools, one setup for each part. I suppose postprocessall will not work in
such situations.

BR
Jan

TimPaterson, 2024-09-08:
Your terminology is different from Fusion ("items in a multibody file") so I'm not sure what you have. PostProcessAll was written because I had a project with a dozen components and over 60 setups which all needed to be updated if a design parameter changed. I don't know of any situation where it wouldn't work. You could export an archive file so I could see what you have.

closed 2025-01-28 by TimPaterson