TimPaterson/Fusion360-Batch-Post #71: Skipping Tool Code

Silver-CNC, 2024-10-02:
Hi Tim,

thanks for creating this great Add-on for Fusion. Since some weeks i´ve the following issue. I have a setup with two different tools, when i run your addon it exports the G-code to the file w/o an issue, but it shows for the first tool just the tool change and not the manufacturing code, for the second tool it shows the movements correctly. Here the output code (with Estlcam postprocessor):
As you can see, for T1 is no G-code exported (seperate G-code for T1 is posted below):

%
(1 Abstandshalter Endschalter)
(T1  D=8. CR=0. - ZMIN=-10. - SCHAFTFRSER)
(T12  D=4. CR=0. - ZMIN=-18. - SCHAFTFRSER)
G90 G94 G17 G91.1
G21
(WHEN USING FUSION FOR PERSONAL USE, THE FEEDRATE OF RAPID)
(MOVES IS REDUCED TO MATCH THE FEEDRATE OF CUTTING MOVES,)
(WHICH CAN INCREASE MACHINING TIME. UNRESTRICTED RAPID MOVES)
(ARE AVAILABLE WITH A FUSION SUBSCRIPTION.)
G53 G0 Z0.
(2D-TASCHE1)
M9
M11
T1 M6
(SCHAFTFRSER 8MM)
S13926 M3
G54
(WHEN USING FUSION FOR PERSONAL USE, THE FEEDRATE OF RAPID)
(MOVES IS REDUCED TO MATCH THE FEEDRATE OF CUTTING MOVES,)
(WHICH CAN INCREASE MACHINING TIME. UNRESTRICTED RAPID MOVES)
(ARE AVAILABLE WITH A FUSION SUBSCRIPTION.)
(2D-TASCHE1 2)
M9 G30
T12 M6
(SCHAFTFRSER 4MM)
S18000 M3
G54
M9
G0 X26.056 Y18.423
G0 G43 Z15. H1
G1 Z5. F350.
G1 Z1. F333.33
G1 Z-1.7 F350.
G1 X26.051 Y18.419 Z-1.804
G1 X26.034 Y18.407 Z-1.907
G1 X26.007 Y18.387 Z-2.006
G1 X25.97 Y18.36 Z-2.1
G1 X25.923 Y18.326 Z-2.187
G1 X25.867 Y18.285 Z-2.266
G1 X25.803 Y18.239 Z-2.335
G1 X25.733 Y18.188 Z-2.393
G1 X25.657 Y18.133 Z-2.439
G1 X25.577 Y18.074 Z-2.473
G1 X25.494 Y18.014 Z-2.493
G1 X25.409 Y17.953 Z-2.5
G3 X23.65 Y14.5 I2.508 J-3.453
G1 Y2.
G2 X21.5 Y-0.15 I-2.15 J0.
G1 X1.5
G2 X-0.65 Y2. I0. J2.15
G1 Y27.
G2 X1.5 Y29.15 I2.15 J0.
G1 X21.5
G2 X23.65 Y27. I0. J-2.15
G1 Y14.5
G2 X21.323 Y11.009 I-3.782 J0.
G3 X18.995 Y7.517 I1.455 J-3.491
G1 Y4.824
G2 X18.676 Y4.505 I-0.319 J0.
G1 X4.324
G2 X4.005 Y4.824 I0. J0.319
G1 Y24.176
G2 X4.324 Y24.495 I0.319 J0.
G1 X18.676
[... 524 lines cut ...]
G3 X-3.469 Y35.741 I0.099 J-1.171
G1 X-3.565 Y32.207
G3 X-2.39 Y31. I1.175 J-0.032
G1 X21.5
G1 X25.389
G3 X25.514 Y31.01 I0. J0.8
G1 X25.606 Y31.025
G1 X25.709 Y31.041 Z-9.993
G1 X25.811 Y31.057 Z-9.973
G1 X25.909 Y31.072 Z-9.939
G1 X26.001 Y31.087 Z-9.893
G1 X26.087 Y31.101 Z-9.835
G1 X26.165 Y31.113 Z-9.766
G1 X26.233 Y31.124 Z-9.687
G1 X26.291 Y31.133 Z-9.6
G1 X26.336 Y31.14 Z-9.506
G1 X26.37 Y31.145 Z-9.407
G1 X26.39 Y31.149 Z-9.304
G1 X26.396 Y31.15 Z-9.2
G1 Z15.
M9
M11
G53 G0 Z0.
M30
%

Silver-CNC, 2024-10-02:
Here the Fusion file with the manufacturing steps.
[Abstandshalter Endschalter v3.zip](https://github.com/user-attachments/files/17232235/Abstandshalter.Endschalter.v3.zip)

Silver-CNC, 2024-10-03:
Hi Tim thanks for your feedback. I checked the Fusion libary, for me the Estlcam postprocessor is already at V44145. When i choose this version, i got the error message:
Tool change G-code (Txx) not found; this post processor is not compatible with Post Process All.

Could you share your version 44104 to test it?

TimPaterson, 2024-10-03:
My reference to estlcam 2.cps was Fusion renaming my attempt to get the latest version, refusing to overwrite the old one. So now that I have this straight, the current version of estlcam.cps is 44145 and it gives the error that it is not compatible.  Comparing the file to version 44104, the line of code that emits the tool change has been deleted. If I edit the current version and add the tool change line copied from old one, it works. I have attached a version with that update:
[estlcam.zip](https://github.com/user-attachments/files/17247552/estlcam.zip)

I don't know what post processor you've been using. The G-code you posted is nothing like what is produced by either the new or old estlcam.cps.

Silver-CNC, 2024-10-04:
Thanks Tim, Postprocess All works with your file. I just did some modifications to change the M00 output to M06 to get back the notifications in Estlcam for a tool change. if you want to update you working table for Estlcam here the Variables:

for Tool change: M9
for ending sequence: M5 M9 (note: Estlcam drives afterwards automatically to the home parking position if defined)

No need to check the "manual tool change" checkbox in the tool libary.

p.s. Do you know how to remove these general comments in the nc output?

"(-Attention- Property Safe Retracts is set to Clearance Height.)
(Ensure the clearance height will clear the part and or fixtures.)
(Raise the Z-axis to a safe height before starting the program.)
(When using Fusion for Personal Use, the feedrate of rapid)
(moves is reduced to match the feedrate of cutting moves,)
(which can increase machining time. Unrestricted rapid moves)
(are available with a Fusion Subscription.)"
[estlcam - funktioniert- okt 2024.zip](https://github.com/user-attachments/files/17261994/estlcam.-.funktioniert-.okt.2024.zip)

closed 2024-10-12 by Silver-CNC