TimPaterson/Fusion360-Batch-Post #74: fusion updated and now post process fails

drrobotnik321, 2024-11-22:
keep getting this error message  "Tool change G-code (Txx) not found; this post processor is not compatible with Post Process All" .  ive been using the same post processor grbl.cps for 2 years and had no issues with post process all until the fusion update last night.

TimPaterson, 2024-11-26:
This failed for me as well when I tested with grbl.cps version 44083.  I was able to get it working by editing the post processor file `grbl.cps` and changing the default value of the `useToolCall` property from `false` to `true`. I found it at line 94 of that file.

Alternatively, the latest version of the post, version 44145, has the default value set to `true`, so you could just switch to that without having to edit it.

drrobotnik321, 2024-11-26:
Interesting.  I found that every worked fine with another CAM document I
was using and determined the only difference between the two was tools I
was using. So I deleted the tools from my library and created them again,
now I don’t have a problem.

MatthiasFraeser, 2024-11-29:
Hello,

I have the same error using the Post Processor for LinuxCNC
It worked fine but after the update I can't post anything.

TimPaterson, 2024-12-01:
I tested with the latest linuxcnc.cps and didn't see a problem.

closed 2025-01-28 by TimPaterson