TimPaterson/Fusion360-Batch-Post #76: Ignores custom post properties

Glennien, 2024-12-15:
Hello!

First off, thank you for this great plugin.

I'm using a custom linuxCNC post thats changes acceleration and G64 values dependant on an operation property.
Its dependant on the correct "Use smoothing" post property (9999), and I have set my default to be assigned to this value.
If the above is fulfilled, it will change acceleration and G64 values according to a operation checkbox property.

This works as intended when posting straight from Fusion360, but when posting via PPA everything seems to be ignored, and there is no smoothing or acceleration change.

Is this a user error, or is there anything in PPA that makes this impossible?

I've attached the custom post(WIP)

[linuxcnc_g.zip](https://github.com/user-attachments/files/18141045/linuxcnc_g.zip)

Thanks!
Sidenote: It would be great if there was an option to only post process 1 operation.

Glennien, 2024-12-15:
This unfortunately seems like a limitation with PPA, as it can only pull defaults - so for now its not compatible with the custom post.

Thanks.

closed 2024-12-15 by Glennien

TimPaterson, 2024-12-16:
I had not heard of operation properties. When I tried your post I got a message saying they were being added. But then I didn't find any way to set them.