TimPaterson/Fusion360-Batch-Post #77: post process all not working

creepyflyer, 2025-01-27:
the error i get 

0 files were written. 1 Setups were skipped due to error:
Failed on setup Setup1: Tool change G-code (Txx) not found; this post processor is not compatible with Post Process All.

the tool change code i have tryed to erase but still the same error

TimPaterson, 2025-01-28:
Are you a new user, or did this used to work? What post processor are you using? Does it put out tool change G-code when you use the post with Fusion?

creepyflyer, 2025-01-28:
> Van "Tim Paterson" ***@***.***>
> Aan "TimPaterson/Fusion360-Batch-Post"
> CC "creepyflyer" ***@***.***>; "Author"
> Datum 28/01/2025 19:07:46
> Onderwerp Re: [TimPaterson/Fusion360-Batch-Post] post process all not
> working (Issue #77)

hey tim
yes new user  and it has never worked    grbl post processor  last 
question i do not understand
i just wanted to get the faster feed rate when the tool is not cutting 
as a private user fusion prohibits this
hope you can help thanks in advance

greetings kurt

TimPaterson, 2025-01-29:
There were some problems with older versions of Fusion's grbl post.  Make sure you're using the latest, version 44161.

creepyflyer, 2025-01-30:
Thank you tim
can you please direct me where to find this version and how to install 
it please searching the internet on grbl post version 44161 gets me 
nowhere

greets kurt

Datum 29/01/2025 20:56:07

TimPaterson, 2025-01-30:
I am not a tutor in Fusion basics. In the Fusion Manufacture workspace, open Post Library from the Manage menu. Select the Fusion library, and type grbl in the search box.  Right click on the post and export it to your computer.  Then when you run PostProcessAll, point it to where you save the post.

creepyflyer, 2025-01-31:
updated the grbl to version  44161
i get the same error

Datum 30/01/2025 19:29:09

gerritv, 2025-01-31:
take a look at this posting, make sure that check box is checked.

zabu83, 2025-02-27:
> ¿Eres un usuario nuevo o esto solía funcionar? ¿Qué postprocesador está utilizando? ¿Emite el código G de cambio de herramienta cuando usa la publicación con Fusion?

I'm having the same problem with the linuxcnc postprocessor. I've tried everything mentioned but I still get the same error. Is there a solution?

creepyflyer, 2025-02-27:
> I'm having the same problem with the linuxcnc postprocessor. I've tried everything mentioned but I still get the same error. Is there a solution?

i gave up much to complicated for me  not worth the hassle   im using this online converter that simple and works https://fenrus75.github.io/FenrusCNCtools/ncopt/ncopt.html

zabu83, 2025-02-27:
> > > ¿Eres un usuario nuevo o esto solía funcionar? ¿Qué postprocesador está utilizando? ¿Emite el código G de cambio de herramienta cuando usa la publicación con Fusion?
> > 
> > 
> > Tengo el mismo problema con el postprocesador linuxcnc. He intentado todo lo mencionado, pero sigo recibiendo el mismo error. ¿Hay alguna solución?
> 
> Renuncié a mucho complicado para mí no vale la pena la molestia estoy usando este convertidor en línea que simple y funciona https://fenrus75.github.io/FenrusCNCtools/ncopt/ncopt.html

I don't know how it works, I load the file but it doesn't do anything.

creepyflyer, 2025-02-27:
> I don't know how it works, I load the file but it doesn't do
its already done it goes very fast just save it an you can use it

tangmi, 2025-04-29:
I ran into this and resolved by unchecking "Manual tool change" in the "Post processor" tab while editing a tool in the tool library. This setting appears to output a (localized) form of:

```gcode
(MANUAL TOOL CHANGE TO Txx)
M0
```

instead of the expected `Txx` command.

TimPaterson, 2025-04-29:
That's a valuable tip I didn't know about.  When you check the `Manual tool change` box with the Tormach post that I use, the `Txx` tool change command is still emitted.  But with GRBL it isn't, resulting in the error.

TimPaterson, 2025-09-26:
This is solved my not checking Manual Tool Change.

closed 2025-09-26 by TimPaterson