TimPaterson/Fusion360-Batch-Post #82: Need a way to set 'use smoothing' to just G64, not G64 with PQ

mazonaught, 2025-05-25:
Title is pretty self explanatory... regular 'post process'/'create nc file' gives the option to choose for 'use smoothing', but your script doesn't give any option at all for 'use smoothing', it just defaults to 'G64 with PQ', which doesn't work with my machine (running PlanetCNC). The only option that works with my machine is the standard 'G64' option all by itself. 

Thanks

TimPaterson, 2025-05-25:
I've never heard of G64, so not really self explanatory, but it certainly has nothing to do with PostProcessAll.  My best guess is that this is some option your post processor has.  If you need a post processor setting that is not the default, then (per issue #51) unfortunately you must edit the post processor code file to make it the default.

mazonaught, 2025-06-04:
It's an option every post processor has.. It's a drop down menu for "use smoothing" in fusions post options. It's not specific to any one post processor, it's just what fusion has it set as by default.   Thanks for your response tho, and for all the work you've put into this project.

P-Braunger, 2025-06-30:
Same Problem to me.
In Fusion360 you can select in the Postprocessing windows an option for smoothing.
I also use PlanetCNC, would be nice to add this option to your Script

[image]

TimPaterson, 2025-07-18:
`PostProcessAll` now uses `NC Programs`, which provides access to all post processor options. Please confirm that v2.0 resolves this issure.

TimPaterson, 2025-09-26:
Assuming this is fixed.

closed 2025-09-26 by TimPaterson