TimPaterson/Fusion360-Batch-Post #87: Latest version not working

gcormier, 2025-08-29:
Saw that I'm way behind so I updated! :)

I did export my first OP sucessfully. Fails on 2nd OP.

[image]

Fusion is otherwise content with my toolpaths it seems

[image]

I reverted to commit ee4e1c159e262661d2fc2320c728eb70227ee7ee and the exports work again.

Note I am using the grblHal post-processor.

TimPaterson, 2025-08-29:
I was able to post the 65 setups in my current project without issue using the GRBL post from the Fusion library, so this is not a widespread problem.  I suggest you also try it with the latest post from the Fusion library.

To troubleshoot you could try to post the offending operation manually using the NC program that PostProcessAll is using.  Open up the program and go to the Operations tab and select only the failing operation, then click Post.

gcormier, 2025-09-04:
Can't repro this so closing - will let you know if I can reliable repro.